Mieze/RTL8111_driver_for_OS_X #4: GA-H81M-S2PV Realtek GbE LAN (VmWare/Parallels second IP address support)

allright, 2015-07-08:
1. Open terminal -> run ping google.com
2. Run any parallels/VMWare virtual machine and set Briged Network (it must use the same network adapter but it will have another MAC/IP for virtual machine).
3. ping -> fails after adding this adapter, and does not restore after closing VMWare/Parallels virtual machine.

The same issue with standard Apple driver: AppleRTL8169Ethernet.kext

Mieze, 2015-07-08:
Sounds more like a routing problem and not like a driver bug. Drivers handle packets on behalf of the network stack, they are the last element in the chain before a packet is handed over to the hardware.

Anyway, you should keep in mind that virtualization software installs its own kexts which intervene into the network system and that closing all virtual machines doesn't unload them. They'll stay in memory until  VMWare is closed completely. With regard to Parallels at least older versions didn't unload their kexts at all.

allright, 2015-07-08:
If I use External  Apple USB Ethernet - everything is OK. So I'm not sure
that it is routing problem.

Mieze, 2015-07-08:
According to my tests version 2.0.0 works fine with VMware. I can't reproduce your issue. With version 1.x.x VMware tends to flood the network subsystem with packets eating up all available buffer memory and effectively killing network connectivity. This is not a driver issue but a weakness of VMware as it seems to rely on the packet scheduler to be present in order to work properly.

allright, 2015-07-09:
I've cloned repo. I'll try to reproduce this bug and will send you debug
logs.

allright, 2015-07-09:
Thanks a lot! Recompiling from sources fix the problem!

Jul  9 11:19:26 Andreys-iMac kernel[0]: AppleUSBEthernet::willTerminate -
Link down, Device 1930e000
Jul  9 11:19:26 Andreys-iMac kernel[0]: Ethernet [AppleUSBEthernet]: Link
down on en1
Jul  9 11:19:26 Andreys-iMac kernel[0]: AppleUSBEthernet::disable - Link
down, Device 1930e000
Jul  9 11:20:36 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]: No
medium selected. Falling back to autonegotiation.
Jul  9 11:20:38 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]: Link up
on en0, 100-Megabit, Full-duplex, flow-control
Jul  9 11:20:43 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:20:43 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:20:44 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:20:45 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:20:53 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:21:01 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:21:05 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:21:22 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:21:32 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:21:48 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:21:49 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
Promiscuous mode enabled. // here STARTS Paralless
Jul  9 11:21:57 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
Promiscuous mode disabled.

Jul  9 11:22:16 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
Promiscuous mode enabled.  // here STARTS Parallesl
Jul  9 11:22:48 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:23:53 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:25:38 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:25:57 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:25:59 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:26:18 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:29:08 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:29:10 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x3, checksums valid: 0x0
Jul  9 11:29:20 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:29:53 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Jul  9 11:29:53 Andreys-iMac kernel[0]: Ethernet [RealtekRTL8111]:
checksums applied: 0x5, checksums valid: 0x0
Ju

ping to google.com does not broken

2015-07-09 11:02 GMT+03:00 Andrey Syvrachev <andreyalright@gmail.com>:

> I've cloned repo. I'll try to reproduce this bug and will send you debug
> logs.
>
>
>> According to my tests version 2.0.0 works fine with VMware. I can't
>> reproduce your issue. With version 1.x.x VMware tends to flood the network
>> subsystem with packets eating up all available buffer memory and
>> effectively killing network connectivity. This is not a driver issue but a
>> weakness of VMware as it seems to rely on the packet scheduler to be
>> present in order to work properly.
>>
>>
>
>
>
> C Уважением,
> Сыврачев Андрей,
> Skype:   andrey.syvrachev,
> Tel:       +79265637695.

closed 2015-07-09 by allright