Mieze/RTL8111_driver_for_OS_X #5: Ling Aggregation/Bonding Not Working

up4, 2015-12-21:
Hello,

This might not be a bug, but I have 2 ethernet adapters on my ASUS X79 Deluxe running Mac OS X 10.11.2:

* Realtek RTL8168G/8111G PCI Express Gigabit Ethernet
* Intel(R) 82579V PCI-E Gigabit

I would like to link them in one "Link Aggregate" virtual interface via Mac OS X's Network System Preferences but when I do that and then click on "advanced", it reports the following (I cannot distinguish between the 2 interfaces, please find attached screenshots):

* "The link state was not valid (i.e. down, half-duplex, or wrong speed)."
* "The port on the switch that the device is connected to doesn't seem to have 802.3ad Link Aggregation enabled."

But the switch does support link aggregation. So I figured it must be something with the drivers. It might not even be possible to aggregate from two different drivers but I thought here might be the best place to ask.

Regards,

Vincent

[image]
[image]

Mieze, 2015-12-21:
Hello Vincent!

Did you configure LACP on the switch? Also make sure you are using the latest development versions of the drivers as they have been built with Xcode 6 and an SDK 10.9 which seems to resolve some strange behavior I noticed on 10.11 with versions built using the SDK 10.8.
You might also take a look at this post: http://www.insanelymac.com/forum/topic/287161-new-driver-for-realtek-rtl8111/?p=2195863

Regards,
Mieze

up4, 2015-12-21:
Hello Mieze. Thanks for your answer and also thanks for porting those drivers.

About the switch: no I didn't. But while you were writing, I did and it got rid of the switch-related message and I am left with the one saying "The link state was not valid (i.e. down, half-duplex, or wrong speed).". I will upgrade all kexts and report back later this week.

On a side note, I'm studying your drivers to optimize my usage but also to see if I could get inspiration to port the 10GBase-T Intel drivers. If you are interested to open another conversation channel, I have a few questions for you on this. My email: vincent@up4.com.

Thanks again,

Vincent

closed 2016-03-13 by Mieze