Mieze/RTL8111_driver_for_OS_X #6: very slow upload transfer speeds that eventually fail...

hoondi, 2016-01-01:
Hi,
Really love your work and am enjoying your RTK2.0 driver on my X58 mobo.

I got my hands on an old GB EP45UD3P which has 2 x RTK 8111C so I installed your driver.

I'm experiencing very slow upload speeds. (download speeds are terrific! > 100MB/sec.)

Mac OS X 10.10.5
C2D @ 3.0GHz
8GB DDR2

I've also connected to the GB EP45UD3P using a 2008 MacPro and SMB. It too is very slow to download anything from the GB EP45UD3P (uploading from the 2008 MacPro to the GB EP45UD3P is very fast though)

[image]

Back on the GB EP45UD3P machine and attempting to upload to my X58 mobo, the transfer starts off quickly, but almost immediately slows down to a crawl of only double digit kilobytes per second.
Eventually, I'll see an error dialog box:
[image]

Console.app shows many lines of:
>1/01/2016 5:58:34.000 pm kernel[0]: smb_iod_reconnect: Reconnected share FILES with server HacProServer._smb._tcp.local
1/01/2016 5:58:35.000 pm kernel[0]: smb_fid_get_kernel_fid: No SMB 2/3 fid found for fid c7ebefe2d8a595de

where the last hexadecimal value is always different.

I'm aware of you mentioning and SMB issue with Broadcom/RTLK Nics but I've also tested with AFP and experience the same issue.

Here are the kexts I'm running:
>Server:~ madmin$ kextstat
Index Refs Address            Size       Wired      Name (Version) <Linked Against>
    1   67 0xffffff7f80a0f000 0x8c50     0x8c50     com.apple.kpi.bsd (14.5.0)
    2    7 0xffffff7f81017000 0x28c0     0x28c0     com.apple.kpi.dsep (14.5.0)
    3   83 0xffffff7f80a18000 0x20500    0x20500    com.apple.kpi.iokit (14.5.0)
    4   89 0xffffff7f80a03000 0xbf50     0xbf50     com.apple.kpi.libkern (14.5.0)
    5   77 0xffffff7f80a00000 0x2d50     0x2d50     com.apple.kpi.mach (14.5.0)
    6   42 0xffffff7f80a3e000 0x8590     0x8590     com.apple.kpi.private (14.5.0)
    7   58 0xffffff7f80a39000 0x4900     0x4900     com.apple.kpi.unsupported (14.5.0)
    8    3 0xffffff7f8101d000 0xa0000    0xa0000    com.apple.kec.corecrypto (1.0) <7 6 5 4 3 1>
    9    0 0xffffff7f81155000 0xc000     0xc000     com.apple.kec.pthread (1) <7 6 5 4 1>
   10    1 0xffffff7f8164a000 0x9000     0x9000     com.apple.kec.Libm (1) <4>
   11   12 0xffffff7f81adf000 0x9000     0x9000     com.apple.iokit.IOACPIFamily (1.4) <7 6 4 3>
   12   20 0xffffff7f80d98000 0x34000    0x34000    com.apple.iokit.IOPCIFamily (2.9) <7 6 5 4 3>
   13    1 0xffffff7f827d2000 0x65000    0x65000    com.apple.driver.AppleACPIPlatform (3.1) <12 11 7 6 5 4 3 1>
   14    0 0xffffff7f81635000 0x2000     0x2000     org.tgwbd.driver.NullCPUPowerManagement (1.0.0d2) <7 4 3>
   15    1 0xffffff7f817fb000 0xb000     0xb000     com.apple.driver.AppleFDEKeyStore (28.30) <8 7 6 5 4 3 1>
   16    1 0xffffff7f81ad8000 0x7000     0x7000     com.apple.iokit.IOReportFamily (31) <5 4 3>
   17    4 0xffffff7f80a47000 0x2e000    0x2e000    com.apple.iokit.IOStorageFamily (2.0) <7 6 5 4 3 1>
   18    0 0xffffff7f81b58000 0x1e000    0x1e000    com.apple.driver.DiskImages (398) <17 7 6 5 4 3 1>
   19    0 0xffffff7f81eea000 0x1d000    0x1d000    org.netkas.driver.FakeSMC (1328) <11 7 5 4 3 1>
   20    1 0xffffff7f823aa000 0xf000     0xf000     com.apple.driver.AppleCredentialManager (1.0) <7 5 4 3 1>
   21    2 0xffffff7f81122000 0xd000     0xd000     com.apple.driver.AppleMobileFileIntegrity (1.0.5) <7 6 5 4 3 2 1>
   22    0 0xffffff7f823b9000 0x23000    0x23000    com.apple.driver.AppleKeyStore (2) <21 20 8 7 6 5 4 3 1>
   23    0 0xffffff7f82437000 0x2b000    0x2b000    com.apple.driver.AppleIntelCPUPowerManagement (218.0.0) <7 6 5 4 3 1>
   24    0 0xffffff7f8101a000 0x2000     0x2000     com.apple.security.TMSafetyNet (8) <7 6 5 4 2 1>
   25    2 0xffffff7f8112f000 0x5000     0x5000     com.apple.kext.AppleMatch (1.0.0d1) <4 1>
   26    1 0xffffff7f81134000 0x17000    0x17000    com.apple.security.sandbox (300.0) <25 21 7 6 5 4 3 2 1>
   27    0 0xffffff7f8114b000 0x9000     0x9000     com.apple.security.quarantine (3) <26 25 7 6 5 4 2 1>
   28    0 0xffffff7f82848000 0x8000     0x8000     com.apple.nke.applicationfirewall (161) <7 6 5 4 3 1>
   29    0 0xffffff7f827a5000 0x3000     0x3000     com.apple.driver.AppleAPIC (1.7) <4 3>
   30    0 0xffffff7f822d3000 0x4000     0x4000     com.apple.driver.AppleSMBIOS (2.1) <7 4 3>
   31    0 0xffffff7f824b6000 0x3000     0x3000     com.apple.driver.AppleHPET (1.8) <11 7 5 4 3>
   32    0 0xffffff7f82301000 0x8000     0x8000     com.apple.driver.AppleRTC (2.0) <11 5 4 3 1>
   33    1 0xffffff7f8252b000 0x5000     0x5000     com.apple.driver.AppleEFIRuntime (2.0) <7 6 5 4 3>
   34    1 0xffffff7f82530000 0xb000     0xb000     com.apple.driver.AppleEFINVRAM (2.0) <33 7 6 5 4 3 1>
   35    2 0xffffff7f81806000 0x8a000    0x8a000    com.apple.iokit.IOHIDFamily (2.0.0) <15 7 6 5 4 3 2 1>
   36    0 0xffffff7f82841000 0x4000     0x4000     com.apple.driver.AppleACPIButtons (3.1) <35 13 11 7 6 5 4 3 1>
   37    6 0xffffff7f81773000 0x71000    0x71000    com.apple.iokit.IOUSBFamily (720.4.4) <12 7 5 4 3 1>
   38    0 0xffffff7f818e2000 0x10000    0x10000    com.apple.driver.AppleUSBUHCI (656.4.1) <37 12 7 5 4 3 1>
   39    0 0xffffff7f81936000 0x19000    0x19000    com.apple.driver.AppleUSBEHCI (705.4.14) <37 12 7 5 4 3 1>
   40    2 0xffffff7f81d46000 0x19000    0x19000    com.apple.iokit.IOAHCIFamily (2.7.5) <5 4 3 1>
[... 403 lines cut ...]
	7 current kernel control sockets
	7 kernel control generation count
	0 bad vendor failures
	74 message too big failures
	0 out of memeory failures
	0 messages dropped due to full socket buffers
	84 message posteds
kctl:
	0 total kernel control modules registered
	10 current kernel control modules registered
	14 current kernel control sockets
	676 kernel control generation count
	340 connection attempts
	0 connection failures
	13 send failures
	0 send list failures
	15 enqueus failures
	15 packets dropped due to full socket buffers
Server:~ madmin$>

I also disabled EEE via the plist which has not helped.

Any suggestions?

Regards,

hoondi, 2016-01-01:
Just letting you know that I installed Lnx2Mac's Realtek RTL81xx Driver and all is well – as expected download and upload speeds.

So that suggests there's nothing obviously wrong with my environment/setup.

Regards,

Mieze, 2016-01-01:
There is something wrong with your setup because you probably have a serious problem with interrupt latency.

closed 2016-01-01 by Mieze