Mieze/RTL8111_driver_for_OS_X #9: Development question

ajmas, 2016-04-19:
Not an issue, but a question:

For this development did you start with an existing Linux driver and then simply port it to the Mac? If so, what were the key pain points? I ask since, I would be interested in seeing what can be done for the Realtek wireless drivers.

Mieze, 2016-04-19:
Yes, I used Realtek's Linux driver as a starting point but rewrote much of the code. With regard to wireless drivers, the problem is that there is no documentations and no include files so that you will have a hard time writing a wireless driver for OS X.

ajmas, 2016-04-19:
The issue is more on the Mac side when you mention lack of documentation or more in the wireless chipset?

Mieze, 2016-04-19:
Both, but the lack of documentation from Apple is the worst.

ajmas, 2016-04-20:
Ok. Thanks for the info.

closed 2016-04-20 by ajmas