Mieze/RTL8111_driver_for_OS_X #11: Very slow upload speeds

mconigliaro, 2017-03-08:
I have a Gigabyte GA-EP45-UD3R with an onboard Realtek 8111C, and since switching to this driver a year or two ago, I've been dealing with very slow upload speeds to the NAS on my LAN (which usually end up timing out). A did some digging and found a bunch of people reporting very similar behavior, but nobody seems to have a solution:

- http://www.insanelymac.com/forum/topic/308416-network-and-streaming-serious-performance-issues-with-clover-and-el-capitan/
- http://www.insanelymac.com/forum/topic/309771-super-slow-uploads-to-shared-folders/
- https://www.tonymacx86.com/threads/slow-lan-speed-el-capitan.174970/
- https://www.reddit.com/r/hackintosh/comments/3tooa0/very_slow_network_transfers/
- Issue #6?

I finally got around to trying [Lnx2Mac's RealtekRTL81xx driver](http://lnx2mac.blogspot.com/p/realtekrtl81xx-osx-driver.html) again, and I've confirmed that it fixes this issue, but I'd prefer to use your driver, since it seems to be under more active development, and it's the only one listed in MultiBeast now. Is there anything I can provide to help troubleshoot this?

Mieze, 2017-03-08:
There is something wrong with interrupt handling or power management of the EP-45 boards which causes this issue. I explained it several times before. Someone with the hardware and the required background knowledge would have to track down the issue but, as I don't have an EP-45 board, it won't be me. Sorry!

closed 2017-03-08 by Mieze

mconigliaro, 2017-03-08:
For what it's worth (and for future Googlers, etc.), I saw some people mentioning that disabling TSO4 and TSO6 helped, so I just opened `/Library/Extensions/RealtekRTL8111.kext/Contents/Info.plist` and changed this section from:

```
<key>enableTSO4</key>
<true/>
<key>enableTSO6</key>
<true/>
```

to:

```
<key>enableTSO4</key>
<false/>
<key>enableTSO6</key>
<false/>
```

Then I ran `sudo touch /System/Library/Extensions && sudo kextcache -u /` to rebuild the kext cache and rebooted, and that seems to have fixed my immediate issue (transferring files to my NAS). I'll keep playing around with this and I'll update this issue if I discover anything else.

Thanks for all your hard work on this driver!