Mieze/RTL8111_driver_for_OS_X #12: 10.12.4

Gromek1982, 2017-04-04:
Hello,

After upgrading to 10.12.4 I started having problems with network. Not 100% sure if it's Kext related since as soon as I log in, my ethernet card is detected, I get an ip address but there's no network available at all. I cannot ping anything (ping is frozen) it's not even trying to ping anything. After few minutes it starts to work and the process repeats. Can anyone help ?

ifamed, 2017-04-04:
Just try use with 10.12.3 😁

Gromek1982, 2017-04-04:
It worked fine with 10.12.3 - problems started after the upgrade and I was just wondering someone can help.

bskendig, 2017-04-10:
@Gromek1982, what network chip do you have, specifically? And is anything relevant logged in the Console?

Mieze, 2017-04-10:
It's definitely not the driver but something in the system must have been messed up, maybe the network configuration or the system caches.

Gromek1982, 2017-04-10:
It turned out it was a Norton Internet Security that was causing this.

Mieze, 2017-04-10:
Funny situation! At least it seems to keep you save (without Internet). ;-)

closed 2017-04-11 by Mieze