Mieze/RTL8111_driver_for_OS_X #13: 8112L is unreliable with this driver under Sierra

bskendig, 2017-04-10:
Just FYI, in case it helps anyone: I tried using this driver under 10.12.x Sierra on a P7P55D-E Pro motherboard with a Realtek 8112L chip, and it was consistently flaky. Time Machine would sometimes transfer up to 100 gigabytes of a new backup, but then it would abruptly drop the connection and fail the backup. No errors in the log (other than Time Machine errors about suddenly not being able to find the backup volume). ssh connections would sometimes abruptly drop. Sometimes I could continue networking in other apps, sometimes networking would be dead until a reboot.

I switched to a USB Ethernet adapter instead, and networking has been rock-solid with that so far.

Mieze, 2017-04-10:
That's the expected behavior because the RTL8112L isn't supported by the driver.

closed 2017-04-10 by Mieze

bskendig, 2017-04-10:
Thank you. I wasn't aware it was a different beast from the 8111.