Mieze/RTL8111_driver_for_OS_X #15: Consistent kernel panics

dignoe, 2017-07-10:
I'm getting kernel panics about once or twice a week using this driver. Below is the output of the Panic Report.

Fri Jul  7 04:22:22 2017

*** Panic Report ***
panic(cpu 0 caller 0xffffff80003ff2dd): Kernel trap at 0xffffff7f82e5b1a2, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff81c14df000, CR3: 0x0000000009072000, CR4: 0x00000000001626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000000, RCX: 0xffffff81c14df000, RDX: 0xffffff81c14ba088
RSP: 0xffffff91ed9b3e40, RBP: 0xffffff91ed9b3e50, RSI: 0x0000000000000000, RDI: 0xffffff81a0beb000
R8:  0x0000595ebe6dc3ac, R9:  0xffffff8025e9cb60, R10: 0x0000000000000000, R11: 0xffffff8000a8a810
R12: 0x0000595ebd3548fc, R13: 0x00000051366cf4a8, R14: 0xffffff81a0beb000, R15: 0xffffff81a0beb2a0
RFL: 0x0000000000010246, RIP: 0xffffff7f82e5b1a2, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff81c14df000, Error code: 0x0000000000000002, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff91ed9b3ad0 : 0xffffff80002ea53c 
0xffffff91ed9b3b50 : 0xffffff80003ff2dd 
0xffffff91ed9b3d30 : 0xffffff800029b593 
0xffffff91ed9b3d50 : 0xffffff7f82e5b1a2 
0xffffff91ed9b3e50 : 0xffffff7f82e5ce39 
0xffffff91ed9b3e70 : 0xffffff7f82e5c7ea 
0xffffff91ed9b3ea0 : 0xffffff7f82e5c09f 
0xffffff91ed9b3ee0 : 0xffffff80008bc69f 
0xffffff91ed9b3f30 : 0xffffff80008baf01 
0xffffff91ed9b3f80 : 0xffffff80008ba4e6 
0xffffff91ed9b3fb0 : 0xffffff80002988f7 
      Kernel Extensions in backtrace:
         com.insanelymac.RealtekRTL8111(2.2.1)[8BA5E6A6-5DCA-3102-B71B-CE613147F452]@0xffffff7f82dfb000->0xffffff7f82e61fff
            dependency: com.apple.iokit.IONetworkingFamily(3.2)[6326DB88-5330-3F0C-91F6-D478AB5E7503]@0xffffff7f816ac000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[BD9FCDEF-30CB-3911-B247-EE51DD069C76]@0xffffff7f80b5e000

BSD process name corresponding to current thread: kernel_task
Boot args: kext-dev-mode=1 rootless=0 slide=0 

Mac OS version:
16F73

Kernel version:
Darwin Kernel Version 16.6.0: Fri Apr 14 16:21:16 PDT 2017; root:xnu-3789.60.24~6/RELEASE_X86_64
Kernel UUID: 
__HIB  text base: 0xffffff8000100000
System model name: iMac13,2

System uptime in nanoseconds: 98263446810472
last loaded kext at 94025360868662: com.apple.driver.AppleXsanScheme	3 (addr 0xffffff7f82fb4000, size 32768)
last unloaded kext at 94808025502246: com.apple.driver.AppleUSBAudio	308.7 (addr 0xffffff7f8301d000, size 364544)
loaded kexts:
com.avira.kext.FileAccessControl	1.2.5
org.virtualbox.kext.VBoxNetAdp	5.0.18
org.virtualbox.kext.VBoxNetFlt	5.0.18
org.virtualbox.kext.VBoxUSB	5.0.18
org.virtualbox.kext.VBoxDrv	5.0.18
com.intel.driver.EnergyDriver	2.0
com.insanelymac.RealtekRTL8111	2.2.1
org.netkas.driver.FakeSMC	1372
com.apple.filesystems.msdosfs	1.10
com.apple.filesystems.udf	2.5
com.apple.filesystems.smbfs	3.1.2
com.apple.driver.AppleBluetoothMultitouch	92
com.apple.filesystems.afpfs	11.0.5
com.apple.nke.asp-tcp	8.0.1
com.apple.filesystems.autofs	3.0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.AGPM	110.23.17
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.AppleHDA	279.48
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.GeForce	10.1.7
com.apple.driver.pmtelemetry	1
com.apple.driver.AppleBacklight	170.9.20
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleMCCSControl	1.3.1
com.apple.driver.AppleIntelFramebufferCapri	10.2.5
com.apple.iokit.IOBluetoothSerialManager	5.0.4f18
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHV	1
com.apple.driver.AppleIntelSlowAdaptiveClocking	4.0.0
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0
com.apple.BootCache	40
com.apple.filesystems.hfs.kext	366.50.19
com.apple.iokit.SCSITaskUserClient	394.50.1
com.apple.iokit.IOAHCIBlockStorage	295.20.1
com.apple.driver.AppleAHCIPort	326.60.1
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleACPIButtons	5.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.driver.AppleIntelCPUPowerManagementClient	219.0.0
com.apple.nke.applicationfirewall	172
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleIntelCPUPowerManagement	219.0.0
com.apple.driver.IOBluetoothHIDDriver	5.0.4f18
com.apple.driver.AppleMultitouchDriver	368.16
com.apple.driver.AppleInputDeviceSupport	76.7
com.apple.security.SecureRemotePassword	1.0
com.apple.iokit.IOUSBUserClient	900.4.1
com.apple.kext.triggers	1.0
com.apple.driver.DspFuncLib	279.48
com.apple.kext.OSvKernDSPLib	525
com.apple.nvidia.driver.NVDAGK100Hal	10.1.7
com.apple.nvidia.driver.NVDAResman	10.1.7
com.apple.driver.AppleBacklightExpert	1.1.0
com.apple.iokit.IOSurface	159.7
com.apple.iokit.IONDRVSupport	516.1
com.apple.driver.X86PlatformPlugin	1.0.0
com.apple.driver.IOPlatformPluginFamily	6.0.0d8
com.apple.driver.AppleHDAController	279.48
com.apple.iokit.IOHDAFamily	279.48
com.apple.iokit.IOAudioFamily	205.15
com.apple.vecLib.kext	1.2.0
com.apple.driver.AppleSMBusPCI	1.0.14d1
com.apple.driver.AppleSMBusController	1.0.18d1
com.apple.iokit.IOSMBusFamily	1.1
com.apple.iokit.IOAcceleratorFamily2	311.13
com.apple.AppleGraphicsDeviceControl	3.14.47
com.apple.iokit.IOGraphicsFamily	515.3
com.apple.iokit.IOSerialFamily	11
com.apple.driver.AppleSMC	3.1.9
com.apple.driver.AppleSSE	1.0
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.4f18
com.apple.iokit.IOBluetoothHostControllerUSBTransport	5.0.4f18
com.apple.iokit.IOBluetoothHostControllerTransport	5.0.4f18
com.apple.iokit.IOBluetoothFamily	5.0.4f18
com.apple.driver.AppleHIDKeyboard	199
com.apple.driver.usb.IOUSBHostHIDDevice	1.1
com.apple.driver.usb.networking	5.0.0
com.apple.driver.usb.AppleUSBHostCompositeDevice	1.1
com.apple.driver.usb.AppleUSBHub	1.1
com.apple.filesystems.hfs.encodings.kext	1
com.apple.driver.AppleUSBMergeNub	900.4.1
com.apple.driver.AppleUSBHostMergeProperties	1.1
com.apple.driver.usb.AppleUSBHostPacketFilter	1.0
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.iokit.IOSCSIMultimediaCommandsDevice	394.50.1
com.apple.iokit.IOBDStorageFamily	1.8
com.apple.iokit.IODVDStorageFamily	1.8
com.apple.iokit.IOCDStorageFamily	1.8
com.apple.iokit.IOAHCISerialATAPI	266.50.1
com.apple.iokit.IOSCSIArchitectureModelFamily	394.50.1
com.apple.iokit.IONetworkingFamily	3.2
com.apple.iokit.IOAHCIFamily	288
com.apple.driver.usb.AppleUSBEHCIPCI	1.1
com.apple.driver.usb.AppleUSBEHCI	1.1
com.apple.driver.usb.AppleUSBXHCIPCI	1.1
com.apple.driver.usb.AppleUSBXHCI	1.1
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.1
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.DiskImages	444.50.16
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0

Mieze, 2017-09-06:
The driver is stable, you need to fix your system. The kernel panics are either a result of a bad UEFI setting (UEFI network stack enabled, etc.), or a problem with the DSDT.

dignoe, 2017-09-06:
I'm not using a DSDT, and the UEFI network stack is disabled in the bios. I will try tweaking my Clover settings. Is there anything that _needs_ to be enabled/disabled in the Clover settings for this to work properly? i.e. the FixLAN_2000 fix is currently checked/true.

Mieze, 2017-09-07:
This doesn't protect you from DSDT errors because the system is using the vanilla DSDT which is usually full of bugs. Also make sure to you have the FixRegions_10000000 patch selected.

dignoe, 2017-09-07:
Ok, thanks. I'll do some digging and experimenting.

closed 2017-09-07 by dignoe

Denisuu, 2019-06-10:
Have you solved this? I'm having the same kernel panics.