Mieze/RTL8111_driver_for_OS_X #16: Slow speeds on GA EP45-UD3L motherboard, Realtek 8111c chipset

stealthdave, 2017-08-02:
I'm getting _very_ slow speeds on a Gigabit network.  Here's the `iperf3` output between the host and the router:
```
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.133 port 64988 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   754 KBytes  6.18 Mbits/sec                  
[  5]   1.00-2.00   sec  1.53 MBytes  12.9 Mbits/sec                  
[  5]   2.00-3.00   sec  2.06 MBytes  17.2 Mbits/sec                  
[  5]   3.00-4.00   sec  2.07 MBytes  17.4 Mbits/sec                  
[  5]   4.00-5.00   sec  2.04 MBytes  17.1 Mbits/sec                  
[  5]   5.00-6.00   sec  2.10 MBytes  17.6 Mbits/sec                  
[  5]   6.00-7.00   sec  2.08 MBytes  17.4 Mbits/sec                  
[  5]   7.00-8.00   sec  2.06 MBytes  17.3 Mbits/sec                  
[  5]   8.00-9.00   sec  2.09 MBytes  17.5 Mbits/sec                  
[  5]   9.00-10.00  sec  2.11 MBytes  17.7 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  18.9 MBytes  15.8 Mbits/sec                  sender
[  5]   0.00-10.00  sec  18.7 MBytes  15.7 Mbits/sec                  receiver
```

System specifications:
Gigabyte EP45-UD3L (Realtek 8111c chipset LAN)
Intel Core2Quad Q6600 (yes, it's REALLY old! :D )
El Capitan 10.11.3
RealtekRTL8111.kext v2.2.1 (from tonymacx86 Multibeast 8.2.3)

On Yosemite, I was able to use lnx2mac's 81xx drivers to achieve gigabit speeds, and I am planning on testing them, they do pre-date El Capitan so I'm unsure of their compatibility.  Please let me know if there are any logs or any other system information that I can provide.

Many thanks for the work you've provided!

stealthdave, 2017-08-03:
Here are the `iperf3` results from lnx2mac's drivers:
```
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.133 port 49576 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  37.0 MBytes   310 Mbits/sec                  
[  5]   1.00-2.00   sec  36.7 MBytes   308 Mbits/sec                  
[  5]   2.00-3.00   sec  36.8 MBytes   308 Mbits/sec                  
[  5]   3.00-4.00   sec  34.9 MBytes   292 Mbits/sec                  
[  5]   4.00-5.00   sec  37.6 MBytes   315 Mbits/sec                  
[  5]   5.00-6.00   sec  35.6 MBytes   299 Mbits/sec                  
[  5]   6.00-7.00   sec  36.0 MBytes   302 Mbits/sec                  
[  5]   7.00-8.00   sec  38.7 MBytes   325 Mbits/sec                  
[  5]   8.00-9.00   sec  35.5 MBytes   298 Mbits/sec                  
[  5]   9.00-10.00  sec  35.9 MBytes   301 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   365 MBytes   306 Mbits/sec                  sender
[  5]   0.00-10.00  sec   364 MBytes   305 Mbits/sec                  receiver
```

Mieze, 2017-08-05:
All Gigabyte EP45 series boards show this strange behavior which is either a power management or an interrupt handling issue. Sorry, but there is nothing I can do for you.

closed 2017-09-06 by Mieze