Mieze/RTL8111_driver_for_OS_X #18: KP on 10.13.6 while downloading

willco007, 2018-07-19:
I had a KP was downloading the Ubuntu 18.04 ISO in Safari from ubuntu.com while AFK. The 2.2.2 version of the RTL driver appears to be part of the stack. System is vanilla install. Below is the stack; what could I provide that would help sort this out?  Thanks.

`
*** Panic Report ***
panic(cpu 0 caller 0xffffff800318776f): Kernel trap at 0xffffff7f851e25b1, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff801b5ddfc0, CR3: 0x000000000bdda000, CR4: 0x00000000001626e0
RAX: 0x0000000000000000, RBX: 0xffffff81a4192000, RCX: 0x0000000000001fc0, RDX: 0x0000000000000100
RSP: 0xffffff81f1813ca0, RBP: 0xffffff81f1813d40, RSI: 0x00000000000001fc, RDI: 0xffffff81a4192000
R8:  0x0000000000000000, R9:  0x0000000000000013, R10: 0x0000000000000000, R11: 0xffffff7f841889b2
R12: 0xffffff81f1813f20, R13: 0xffffff801b5dc000, R14: 0xffffff81a4192000, R15: 0xffffff81a4192000
RFL: 0x0000000000010206, RIP: 0xffffff7f851e25b1, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff801b5ddfc0, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff81f1813770 : 0xffffff800306c1c6 
0xffffff81f18137c0 : 0xffffff8003195274 
0xffffff81f1813800 : 0xffffff8003187544 
0xffffff81f1813870 : 0xffffff800301e1e0 
0xffffff81f1813890 : 0xffffff800306bc3c 
0xffffff81f18139c0 : 0xffffff800306b9fc 
0xffffff81f1813a20 : 0xffffff800318776f 
0xffffff81f1813b90 : 0xffffff800301e1e0 
0xffffff81f1813bb0 : 0xffffff7f851e25b1 
0xffffff81f1813d40 : 0xffffff7f851e2362 
0xffffff81f1813d60 : 0xffffff7f845fa4ce 
0xffffff81f1813d90 : 0xffffff800369ad68 
0xffffff81f1813e00 : 0xffffff7f845fa538 
0xffffff81f1813e60 : 0xffffff7f845fe7ac 
0xffffff81f1813ee0 : 0xffffff800330a665 
0xffffff81f1813fa0 : 0xffffff800301d557 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[5A99D802-B477-3D83-9F76-304605479B9F]@0xffffff7f845f0000->0xffffff7f8461ffff
         com.insanelymac.RealtekRTL8111(2.2.2)[24D129C7-C25C-3D7B-9E10-EA71D9BE6511]@0xffffff7f85183000->0xffffff7f851e9fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[5A99D802-B477-3D83-9F76-304605479B9F]@0xffffff7f845f0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[AEA4C7F5-CCB0-338B-B471-CF28A9792522]@0xffffff7f83934000

BSD process name corresponding to current thread: kernel_task
`

Mieze, 2018-07-19:
This is not a driver problem but a result of a messed up system (system caches, DSDT related issue, wrong BIOS setting, etc.) which can be hard to track down.

willco007, 2018-07-19:
I have had no issues with this system for 5+ years, including the ethernet driver.  I don't have a custom DSDT, nor have I changed any BIOS settings in years. I can clear system caches, but that seems like a long shot.  It could very well be an OS one-off issue; but give the ethernet driver was in the stack I thought I'd log it.

Mieze, 2018-07-19:
Driver caused kernel panics occur when a driver fails to communicate with the OS correctly but the driver interface is stable and the driver uses it in the way it should so that I'm quite confident that there is no driver bug which causes KPs, at least there hasn't been such a problem with any of my drivers during the last 4 years.

willco007, 2018-07-19:
Thanks for the replies, I'll keep an eye on it. Thanks again.

freite, 2019-07-24:
not sure if this is exactly the same problem or not, but macos crashes when sleep is activated sometimes and it seems related to this driver. the kernel panic is reproducible every time when the nic is under heavy load, coping a large file using scp for example. 

[crash-RealtekRTL8111.txt](https://github.com/Mieze/RTL8111_driver_for_OS_X/files/3427954/crash-RealtekRTL8111.txt)

mugukamil, 2019-09-23:
I have that issue too, When my PC wanted go to sleep, I recieved KP, and reboot

`*** Panic Report ***
panic(cpu 1 caller 0xffffff80004dbadd): Kernel trap at 0xffffff7f80f5970b, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff801cd362b0, CR3: 0x000000000db9a000, CR4: 0x00000000001626e0
RAX: 0x0000000000000000, RBX: 0xffffff81a5b1f000, RCX: 0x00000000000012b0, RDX: 0x0000000000000100
RSP: 0xffffff81f5b43cb0, RBP: 0xffffff81f5b43d50, RSI: 0x000000000000012b, RDI: 0xffffff81a5b1f000
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0xffffff8000df2cc0
R12: 0xffffff81f5b43f20, R13: 0xffffff801cd35000, R14: 0xffffff81a5b1f000, R15: 0xffffff81a5b1f000
RFL: 0x0000000000010202, RIP: 0xffffff7f80f5970b, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff801cd362b0, Error code: 0x0000000000000000, Fault CPU: 0x1, PL: 0, VF: 1

Backtrace (CPU 1), Frame : Return Address
0xffffff81f5b43780 : 0xffffff80003ae6ed 
0xffffff81f5b437d0 : 0xffffff80004ea185 
0xffffff81f5b43810 : 0xffffff80004db8ba 
0xffffff81f5b43880 : 0xffffff800035bb40 
0xffffff81f5b438a0 : 0xffffff80003ae107 
0xffffff81f5b439c0 : 0xffffff80003adf53 
0xffffff81f5b43a30 : 0xffffff80004dbadd 
0xffffff81f5b43ba0 : 0xffffff800035bb40 
0xffffff81f5b43bc0 : 0xffffff7f80f5970b 
0xffffff81f5b43d50 : 0xffffff7f80f594c0 
0xffffff81f5b43d70 : 0xffffff7f80ecd558 
0xffffff81f5b43da0 : 0xffffff8000a5a908 
0xffffff81f5b43e10 : 0xffffff7f80ecd5c2 
0xffffff81f5b43e60 : 0xffffff7f80ed1837 
0xffffff81f5b43ee0 : 0xffffff8000672ac1 
0xffffff81f5b43fa0 : 0xffffff800035b0ce 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[E04C914F-B8A8-3EBF-8D4A-E492DAD7A657]@0xffffff7f80ec3000->0xffffff7f80ef2fff
         com.insanelymac.RealtekRTL8111(2.2.2)[6AB67628-5BB1-32EB-8881-4FBD58E3AFBB]@0xffffff7f80efa000->0xffffff7f80f60fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[E04C914F-B8A8-3EBF-8D4A-E492DAD7A657]@0xffffff7f80ec3000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[BFDEB8D4-50FE-3DDB-87B7-F6A504393830]@0xffffff7f80c95000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 slide=0 

`

BenTalagan, 2019-11-22:
Have seen this too, multiple times, lately, under catalina, while the system tries to go to sleep but the network is under heavy load (large file downloads, etc).

kocoman1, 2020-06-06:
Disabled aspm but still panics

`panic(cpu 3 caller 0xffffff800644c32a): Kernel trap at 0xffffff7f892b5edb, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff81875134e0, CR3: 0x000000001b702000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000081, RCX: 0x00000000000014e0, RDX: 0x0000000000000200
RSP: 0xffffff81874bbe10, RBP: 0xffffff81874bbea0, RSI: 0xffffff803048aee0, RDI: 0x000000000000014e
R8:  0x0000000000000000, R9:  0x0000000000989680, R10: 0xffffff8006df5c40, R11: 0xffffff8006df5280
R12: 0xffffff80302df000, R13: 0xffffff8187512000, R14: 0x0000000000000000, R15: 0x0000000000000001
RFL: 0x0000000000010202, RIP: 0xffffff7f892b5edb, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff81875134e0, Error code: 0x0000000000000000, Fault CPU: 0x3, PL: 0, VF: 1

Backtrace (CPU 3), Frame : Return Address
0xffffff81874bb730 : 0xffffff80063215cd mach_kernel : _handle_debugger_trap + 0x49d
0xffffff81874bb780 : 0xffffff800645a3c5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81874bb7c0 : 0xffffff800644bf7e mach_kernel : _kernel_trap + 0x4ee
0xffffff81874bb810 : 0xffffff7f8919aa7d as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x46d
0xffffff81874bb8b0 : 0xffffff80062c7a40 mach_kernel : _return_from_trap + 0xe0
0xffffff81874bb8d0 : 0xffffff8006320c97 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff81874bb9d0 : 0xffffff8006321087 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff81874bba20 : 0xffffff8006ac2c7c mach_kernel : _panic + 0x54
0xffffff81874bba90 : 0xffffff800644c32a mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff81874bbc10 : 0xffffff800644c028 mach_kernel : _kernel_trap + 0x598
0xffffff81874bbc60 : 0xffffff7f8919aa7d as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x46d
0xffffff81874bbd00 : 0xffffff80062c7a40 mach_kernel : _return_from_trap + 0xe0
0xffffff81874bbd20 : 0xffffff7f892b5edb com.insanelymac.RealtekRTL8111 : __ZN7RTL811111rxInterruptEP18IONetworkInterfacejP11IOMbufQueuePv + 0x43
0xffffff81874bbea0 : 0xffffff7f892b92d4 com.insanelymac.RealtekRTL8111 : __ZN7RTL811117interruptOccurredEP8OSObjectP22IOInterruptEventSourcei + 0xb8
0xffffff81874bbee0 : 0xffffff8006a2eb6d mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x17d
0xffffff81874bbf30 : 0xffffff8006a2d43e mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x11e
0xffffff81874bbf70 : 0xffffff8006a2ca36 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x36
0xffffff81874bbfa0 : 0xffffff80062c713e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         as.vit9696.VirtualSMC(1.1.4)[D51ACAF0-8CAA-3DC2-8260-C51D354709DF]@0xffffff7f8918b000->0xffffff7f891b1fff
            dependency: as.vit9696.Lilu(1.4.5)[04010205-4C3D-3B04-BAF5-48AA69FE71F2]@0xffffff7f89108000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9D1FF279-C4A2-3344-902F-E0B22B508689]@0xffffff7f874a7000
         com.insanelymac.RealtekRTL8111(2.2.1)[07D64653-7867-3C3A-837B-E491B33C21B7]@0xffffff7f89256000->0xffffff7f892c4fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[3F6EDFDE-6EF7-38D8-8006-C4EE6DB4D4B5]@0xffffff7f86c65000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f86d6a000

BSD process name corresponding to current thread: kernel_task
Boot args: igfxframe=0x19120000 -v keepsyms=1

Mac OS version:
19E266

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64
Kernel UUID: AB0AA7EE-3D03-3C21-91AD-5719D79D7AF6
Kernel slide:     0x0000000006000000
Kernel text base: 0xffffff8006200000
__HIB  text base: 0xffffff8006100000
System model name: iMac17,1 (Mac-DB15BD556843C820)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 4708127040438
last loaded kext at 145584203313: @filesystems.smbfs	3.4.2 (addr 0xffffff7f893d0000, size 450560)
loaded kexts:
org.virtualbox.kext.VBoxNetAdp	6.1.8
org.virtualbox.kext.VBoxNetFlt	6.1.8
org.virtualbox.kext.VBoxUSB	6.1.8
org.virtualbox.kext.VBoxDrv	6.1.8
com.paragon-software.filesystems.extfs	30.3.11
com.paragon-software.lvm-for-mac	1
com.insanelymac.RealtekRTL8111	2.2.1
as.vit9696.VirtualSMC	1.1.4
as.vit9696.WhateverGreen	1.4.0
com.rehabman.driver.USBInjectAll	0.6.2
as.vit9696.Lilu	1.4.5
@filesystems.smbfs	3.4.2
>!AUpstreamUserClient	3.6.8
[... 91 lines cut ...]
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.101.3
|IOSCSIArchitectureModel!F	422.101.1
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	840.3
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

`

closed 2020-07-13 by Mieze

willco007, 2020-12-18:
I still continue to see this KP occasionally downloading/uploading while AFK on 10.14.x and 10.15.x years after the initial report and on different machines. I've finally decided to get a native ethernet card to avoid using this driver, which is too bad because it works great otherwise.