Mieze/RTL8111_driver_for_OS_X #19: Self assigned IP on Mojave

qBor, 2018-10-12:
This kext doesn’t work correctly on Mojave 
Always « self assigned IP »

melvyn2, 2018-10-18:
I have been having this issue on 10.14 too. I will gladly upload kernel logs if you tell me how. I've checked with different cables (doesn't work), different devices on the other end (doesn't work), and different computers (works).

Mieze, 2018-10-18:
Although I haven't tried it with Mojave myself, there is no reason to believe that there is a driver problem with the new OS. Anyway, I added some troubleshooting instructions to the Readme, which also explain how to retrieve kernel logs.

melvyn2, 2018-10-19:
Here is the kernel log:

[Terminal Saved Output.txt.zip](https://github.com/Mieze/RTL8111_driver_for_OS_X/files/2496180/Terminal.Saved.Output.txt.zip)

Mieze, 2018-10-19:
Please use the debug version of the driver in order to collect kernel logs as there is almost no valuable information in your log file.

melvyn2, 2018-10-19:
[log.txt.zip](https://github.com/Mieze/RTL8111_driver_for_OS_X/files/2497055/log.txt.zip)
Here you go! Sorry about that, I didn't realize that the debug version was needed.

Mieze, 2018-10-19:
The driver is working as expected...

2018-10-06 10:32:06.816943-0700 0x118      Default     0x0                  0      0    kernel: (RealtekRTL8111) Ethernet [RealtekRTL8111]: Link up on en0, 10-Megabit, Full-duplex, No flow-control

 ...but I think you've got a messed up network configuration.

Mieze, 2018-10-19:
Take a closer look at your logs. The Ethernet port as well as Airport are both named en0 so that there must be something wrong with the config and it's no wonder that it isn't working.

melvyn2, 2018-10-19:
DPCIManager and `ifconfig` both report the WDN4800 on `en1` and the Realtek card on `en0`...

Mieze, 2018-10-19:
> DPCIManager and `ifconfig` both report the WDN4800 on `en1` and the Realtek card on `en0`...

Then you should find out why there is this line in the logs

2018-10-17 07:30:19.232001-0700 0xf6       Default     0x0                  0      0    kernel: (IO80211Family) AirPort: Link Up on en0

melvyn2, 2018-10-19:
Ok, I'll see if I can fix that.

melvyn2, 2018-10-19:
After clearing `/Library/Preferences/SystemConfiguration/NetworkInterfaces.plist11` and a reboot, the WDN4800 is on `en0` and the Realtek card on `en1`, both after boot and in the logs.

Mieze, 2018-10-19:
Is the problem solved?

melvyn2, 2018-10-19:
No, it isn't. (should've mentioned)

Mieze, 2018-10-19:
What does the interface statistics (netstat -I en1) say?

Mieze, 2018-10-19:
...and what's the result of "ifconfig -v"?

melvyn2, 2018-10-19:
```
$ netstat -I en1
Name  Mtu   Network       Address            Ipkts Ierrs    Opkts Oerrs  Coll
en1   1500  <Link#5>    40:8d:5c:4d:a3:9c        6     0      243     0     0
en1   1500  melvyns-iha fe80:5::1c74:bd6c        6     -      243     -     -
en1   1500  169.254       melvyns-ihac.lo        6     -      243     -     -
```

melvyn2, 2018-10-19:
```
$ ifconfig -v
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384 index 1
	eflags=12000000<ECN_DISABLE,SENDLIST>
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000 
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	nd6 options=201<PERFORMNUD,DAD>
	link quality: 100 (good)
	state availability: 0 (true)
	timestamp: disabled
	qosmarking enabled: no mode: none
	low power mode: disabled
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280 index 2
	eflags=1000000<ECN_ENABLE>
	state availability: 0 (true)
	qosmarking enabled: no mode: none
	low power mode: disabled
stf0: flags=0<> mtu 1280 index 3
	eflags=1000000<ECN_ENABLE>
	state availability: 0 (true)
	qosmarking enabled: no mode: none
	low power mode: disabled
XHC20: flags=0<> mtu 0 index 4
	eflags=41000000<ECN_ENABLE,FASTLN_ON>
	state availability: 0 (true)
	qosmarking enabled: yes mode: none
	low power mode: disabled
en1: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500 index 5
	eflags=410009c1<AUTOCONFIGURING,ACCEPT_RTADV,TXSTART,RXPOLL,ARPLL,ECN_ENABLE,FASTLN_ON>
	options=6b<RXCSUM,TXCSUM,VLAN_HWTAGGING,TSO4,TSO6>
	ether 40:8d:5c:4d:a3:9c 
	inet6 fe80::1c74:bd6c:ec79:9897%en1 prefixlen 64 secured scopeid 0x5 
	inet 169.254.243.94 netmask 0xffff0000 broadcast 169.254.255.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (10baseT/UTP <full-duplex>)
	status: active
	type: Ethernet
	link quality: 100 (good)
	state availability: 0 (true)
	scheduler: FQ_CODEL 
	link rate: 10.00 Mbps
	qosmarking enabled: yes mode: none
	low power mode: disabled
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500 index 6
	eflags=412008c0<ACCEPT_RTADV,TXSTART,ARPLL,NOACKPRI,ECN_ENABLE,FASTLN_ON>
	ether ec:08:6b:79:9c:d8 
	inet6 fe80::1476:582d:1c5e:8135%en0 prefixlen 64 secured scopeid 0x6 
	inet 10.0.1.15 netmask 0xffffff00 broadcast 10.0.1.255
	inet6 2601:647:5401:145c:8de:bc60:3d4a:3be prefixlen 64 autoconf secured 
	inet6 2601:647:5401:145c:fd90:3102:78a5:4652 prefixlen 64 autoconf temporary 
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
	type: Wi-Fi
	state availability: 0 (true)
	scheduler: FQ_CODEL (driver managed)
	link rate: 21.60 Mbps
	qosmarking enabled: yes mode: none
	low power mode: disabled
p2p0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> mtu 2304 index 7
	eflags=41000080<TXSTART,ECN_ENABLE,FASTLN_ON>
	ether 0e:08:6b:79:9c:d8 
	media: autoselect
	status: active
	type: Wi-Fi
	state availability: 0 (true)
	scheduler: FQ_CODEL (driver managed)
	link rate: 10.00 Mbps
	qosmarking enabled: yes mode: none
	low power mode: disabled
utun0: flags=8051<UP,POINTOPOINT,RUNNING,MULTICAST> mtu 2000 index 8
	eflags=1002080<TXSTART,NOAUTOIPV6LL,ECN_ENABLE>
	inet6 fe80::5f50:e54c:eb6c:e136%utun0 prefixlen 64 scopeid 0x8 
	nd6 options=201<PERFORMNUD,DAD>
	agent domain:ids501 type:clientchannel flags:0xc3 desc:"IDSNexusAgent ids501 : clientchannel"
	state availability: 0 (true)
	scheduler: FQ_CODEL 
	qosmarking enabled: no mode: none
	low power mode: disabled
```

Mieze, 2018-10-19:
On the driver level everything is working fine, the link is established and packet reception/transmission is working. The problem must be somewhere else. Maybe a DHCP, routing or firewall related problem?

melvyn2, 2018-10-19:
I'll go check if it works on linux...

melvyn2, 2018-10-19:
Linux and windows aren't working either, so this is probably a hardware problem.

Mieze, 2018-10-19:
Sorry to hear that!

closed 2020-07-13 by Mieze