Mieze/RTL8111_driver_for_OS_X #21: Tx timeout. Lost interrupt? Tx stalled? Resetting chipset

cattyhouse, 2019-03-07:
Hi Mieze

I sometimes got random ethernet lost, so today i performed iperf3 test to see if this will happen again, and i caught it, 

**after below iperf3 lan test:**

[ ID] Interval           Transfer     Bitrate
[  7]   0.00-690.94 sec  74.1 GBytes   921 Mbits/sec                  sender
[  7]   0.00-690.94 sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated

**i got below error:**

`this part is boot message: `

2019-03-07 13:47:01.824101+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: PCIe ASPM support disabled.
2019-03-07 13:47:01.824106+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: EEE support enabled.
2019-03-07 13:47:01.824111+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: RxPoll support enabled.
2019-03-07 13:47:01.824125+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: TCP/IPv4 segmentation offload enabled.
2019-03-07 13:47:01.824131+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: TCP/IPv6 segmentation offload enabled.
2019-03-07 13:47:01.824137+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: TCP/IPv6 checksum offload enabled.
2019-03-07 13:47:01.824150+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Version 2.2.2 using interrupt mitigate value 0x5f51. Please don't support tonymacx86.com!
2019-03-07 13:47:01.824170+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: PCI power management capabilities: 0xffc3.
2019-03-07 13:47:01.824173+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: PME# from D3 (cold) supported.
2019-03-07 13:47:01.824189+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: PCIe link capabilities: 0x00077c11, link control: 0x0000.
2019-03-07 13:47:01.824191+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Disable PCIe ASPM.
2019-03-07 13:47:01.847100+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: RTL8168E-VL/8111E-VL: (Chipset 16) at 0x<private>, 94:de:80:a9:e8:1d
2019-03-07 13:47:01.847875+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Device is WoL capable.
2019-03-07 13:47:01.847898+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: MSI interrupt index: 1
2019-03-07 13:47:01.848140+0800 0xde       Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: kIOEthernetWakeOnMagicPacket added to filters.
2019-03-07 13:47:01.933295+0800 0x105      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Already in power state 1.
2019-03-07 13:47:08.234436+0800 0x284      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: No medium selected. Falling back to autonegotiation.
2019-03-07 13:47:10.608235+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: pollIntervalTime: 170us
2019-03-07 13:47:10.608242+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Link up on en0, 1-Gigabit, Full-duplex, No flow-control, EEE
2019-03-07 13:47:36.230599+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) 

`this part i changed advanced settings to manual`

Ethernet [RealtekRTL8111]: Link down on en0
2019-03-07 13:47:38.481940+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: pollIntervalTime: 170us
2019-03-07 13:47:38.481961+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Link up on en0, 1-Gigabit, Full-duplex, flow-control, EEE
2019-03-07 14:11:26.101284+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) 

`this part i caught the error after 74GB of iperf3 test`

Ethernet [RealtekRTL8111]: Tx timeout. Lost interrupt?
2019-03-07 14:11:27.101353+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[97]: opts1=0x20000000, opts2=0x60000000, addr=0x826c8aa.
2019-03-07 14:11:27.101365+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[98]: opts1=0x900005a8, opts2=0x60000000, addr=0x84d7798.
2019-03-07 14:11:27.101374+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[99]: opts1=0xa0000042, opts2=0x60000000, addr=0x8362baa.
2019-03-07 14:11:27.101383+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[100]: opts1=0x900005a8, opts2=0x60000000, addr=0x85e5000.
2019-03-07 14:11:27.101392+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[101]: opts1=0xa0000042, opts2=0x60000000, addr=0x82dc0aa.
2019-03-07 14:11:27.101400+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[102]: opts1=0x900004a4, opts2=0x60000000, addr=0x85e5000.
2019-03-07 14:11:27.101409+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[103]: opts1=0xa0000042, opts2=0x60000000, addr=0x817d9aa.
2019-03-07 14:11:27.101418+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[104]: opts1=0x900004a4, opts2=0x60000000, addr=0x85e5000.
2019-03-07 14:11:27.101427+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[105]: opts1=0xa0000042, opts2=0x60000000, addr=0x81847aa.
2019-03-07 14:11:27.101435+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: desc[106]: opts1=0x900004a4, opts2=0x60000000, addr=0x85e5000.
2019-03-07 14:11:27.101454+0800 0x3d6c     Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Tx stalled? Resetting chipset. ISR=0xc0, IMR=0x803f.
2019-03-07 14:11:29.448594+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: pollIntervalTime: 170us
2019-03-07 14:11:29.448611+0800 0x12c      Default     0x0                  0      0    kernel: (kernel) Ethernet [RealtekRTL8111]: Link up on en0, 1-Gigabit, Full-duplex, flow-control, EEE

cattyhouse, 2019-03-07:
now i installed the kext to /L/E, and it seems i no longer get this error after  128 GBytes of iperf3 transfer.

cattyhouse, 2019-03-16:
still the same, and this kext can also cause kernel panic, sleep turns into restart.

Mieze, 2019-03-17:
You've got a messed up system. Delete system caches, reboot and then recreate the kernel cache in order to resolve the issue.

cattyhouse, 2019-03-17:
How do i delete system cache?

Mieze, 2019-03-17:
Google is your friend! I don't want to answer this question over and over again as it's a common question.

cattyhouse, 2019-03-17:
> Google is your friend! I don't want to answer this question over and over again as it's a common question.

Thanks, i know how to clear the kext cache if that's what you mean, "sudo kextcache -i / ", but what other cache do i have to clear? ~/Library/Caches or /Library/Caches or even /System/Library/Caches ? 

plus i don't have any kexts in LE or SLE

Mieze, 2019-03-17:
Goto /System/Library/Caches and delete all the files from the directory and it's subdirectories but DON'T delete the directories itself. Also DON'T delete any aliases inside the directories.

cattyhouse, 2019-03-17:
Thank you ! I will keep an eye on it, will let you know.

cattyhouse, 2019-03-18:
still the sam:

issue one: randomly Rx Stalled, network disconnected, have to unpugin and plugin the cable
issue two: randomly sleep turns into reboot.

here is what i did to delete caches:

sudo rm -rf ~/Library/Caches/*
sudo rm -rf /Library/Caches/*
sudo rm -rf /System/Library/Caches/*

Reboot

sudo kextcache -system-prelinked-kernel
sudo kextcache -i /
sudo kextcache -system-cache
sudo update_dyld_shared_cache -force

Reboot again

cd /Library/Preferences/SystemConfiguration
sudo rm -rf NetworkInterfaces.plist com.apple.airport.preferences.plist preferences.plist com.apple.wifi.message-tracer.plist com.apple.smb.server.plist com.apple.AutoWake.plist com.apple.Boot.plist com.apple.network.eapolclient.configuration.plist

Reboot again

after those steps. the 2 issues still remain.

closed 2020-07-09 by cattyhouse

ledge74, 2020-12-20:
@cattyhouse did you manage to solve the first issue of network disconnected ?