Mieze/RTL8111_driver_for_OS_X #22: 1000baseT but only getting 100m throughput

pingram3541, 2019-03-25:
Hi Mieze, thx for making this driver alternative available to us!

I'm still on v 2.2.1 and thought I'd reach out before updating to 2.2.2 as I'm not sure the fixes apply to my problem.  On the same rig running Linux I get gigabit speeds w/ the same adaptor, cable and switch port but booting into OSX 10.12.5 it never exceeds 100m even though it is set to 1000baseT full-duplex w/ default MTU.  Any ideas how I can troubleshoot/fix this?

requeijaum, 2019-04-20:
I think that I'm having the same problem.
Still: I don't know which version of the driver I'm running on my laptop with El Capitan.

str3g, 2019-10-28:
Same problem! Adapter 8111h has anyone found how to fix the problem? Thanks!

k00l3, 2020-01-26:
I've been experiencing the exact same issue. installed MacOS 10.12.5  and used Version 2.2.1  and then upgraded to 2.2.2 but still only getting under 100Mbps throughput even thou 1Gbps is reported.

k00l3, 2020-01-28:
I managed to get it to work properly if anyone is interested .

I removed all RealtekRTLXXX.kext I could find as well as AppleRTL8169Ethernet.kext in /L/E & /S/L/E rebuilt and repaired cache.

Downloaded the latest Kext from releases installed, rebuilt and repaired cache.Rebooted and Boom!  1Gbps speeds.

life1347, 2020-05-06:
@k00l3 save my day! thank you so much.
So in my case, there is a `AppleRTL8169Ethernet.kext` in 

```
/System/Library/Extensions/IONetworkingFamily.kext/Contents/PlugIns/
```

Remove it, replaced old `RealtekRTLXXX.kext` with the new one, run
```
sudo kextcache -i /
```

and it works like a charm.

closed 2020-07-13 by Mieze