Mieze/RTL8111_driver_for_OS_X #26: does this work for realtek usb wifi adapters

myusrn, 2020-03-15:
Is this RealtekRTL8111 kext only going to work for realtek motherboard integrated ethernet with that specific model number or might it also work for the tp-link archer t2u nano and wn725n usb wifi adapters given the latter shows up in windows device manager as a "Realtek RTL8188EU Wireless LAN 802.1 In USB 2.0 Network Adapter" ?

chromer030, 2020-03-16:
For Realtek USB Wireless devices use this :

https://github.com/chris1111/Wireless-USB-Adapter

myusrn, 2020-03-26:
Thanks for the pointer.  I tried it on a clean system with no prior tp-line/realtek usb wifi adapter drivers installed. That repo contains a "WIFI Icon Collor.pkg" which appears to just change the color of the wifi icon on the status bar.  It doesn't provide a driver for my realtek / tp-link usb wifi adapters.  Am i overlooking something?

chromer030, 2020-03-27:
The packages in release tab  contains Drivers and utility :
https://github.com/chris1111/Wireless-USB-Adapter/files/4301778/Wireless.USB.Adapter-V11.zip
WIFI Icon Collor.pkg just is diffirents status bar icons color.

Wireless.USB.Adapter-V11.zip contains :

[image]

myusrn, 2020-03-27:
Thanks for the clarifications and additional details.  Would you know if one should be able to extract the .kext folders from the Wireless USB Adapter-V11.pkg, using whatever utility one uses to crack open pkg files, and place them in an opencore efi system partitions EFI\OC\Kexts folder and expect that a realtek based tl-link archer t2u nano or wn725n usb adapter would get enabled for use in the opencore booted macOS recovery environment?

chromer030, 2020-03-27:
The kexts inside Wireless.USB.Adapter-V11.zip are original and signed by Realtek , so you can install them in Library/extension via installer or manually and use them with fully enabled SIP.
The issue is not kext ! , all is about utility that use kexts and scan for networks and connects to them.
So if you just extract and install kexts , the usb adapter will be recognized to macOS but you cant connect to any network because the lack of Realtek Utility !
in macOS recovery you should try to load kexts via OpenCore , then via Terminal try to load Realtek Utility manually maybe it's works and you can use your USB wifi Adapter , i didn't tried this method but maybe works.

For inspecting and extracting contents of PKG files you can use Pacifist :
https://www.charlessoft.com/

myusrn, 2020-03-28:
Just tested that and it works.  

In my opencore BOOT volume's /EFI/OC/Drivers i added RtWlanDisk.kext, RtWlanU.kext, RtWlanU1827.kext acquired from the pacifist unpacking of Wireless.USB.Adapter-V11.zip contained Wireless.USB.Adapter-V11.pkg with kexts located in '\choice41 Folder\choice44 Folder\drivers Folder'.   I updated opencore BOOT volume's /EFI/OC/config.plist to load those kexts using propertree-master\PropertyTree.bat | File | OC Clean Snapshot.  

Then I copied the '\choice41 Folder\statusbarapp Folder' to the opencore BOOT volume's root for access when i'm booted into the macOS recovery environment.  I then booted into that environment, hoped into a terminal window and at the '-bash-3.2#' prompt entered  /Volumes/BOOT/statusbarapp\ Folder/StatusBarApp.app/Contents/MacOS/StatusBarApp which launched the status bar utility used to pick a wifi network, enter the credentials, and connect.  I wasn't clear on how i could launch that program to leave it running when i killed the terminal window to go back to the main recovery window to access macOS install/recover process but it didn't appear to matter since the network connection kept running even when the StatusBarApp had been closed.

chromer030, 2020-03-28:
For detaching any process or application from terminal , so after that you can terminate Terminal without terminating the executed process you can use following commands :

nohup /Volumes/BOOT/statusbarapp\ Folder/StatusBarApp.app/Contents/MacOS/StatusBarApp  &

or simply :

open  /Volumes/BOOT/statusbarapp\ Folder/StatusBarApp.app

Thank you very very much for reporting results.
Wish you all the best.

myusrn, 2020-03-28:
Thanks for the suggestions on options to enable keeping the StatusBarApp running when i exit out of the terminal session.  I just tested them in the opencore booted macOS 10.15.4 recovery environment's terminal bash shell.  Both 'nohup' and 'open' produce 'command not found' results so fortunately the realtek based usb wifi adapter keeps working once network connection has been established and StatusBarApp is killed off.

chromer030, 2020-03-28:
sorry for non operating commands , it seemed that Apple doesn't provide essential commands on Recovery while they are working perfect under normal macOS.
Perhaps if we locate binary path of nohup or open on macOS partition or copy this binaries on Opencore root directory , we could use them.

myusrn, 2020-03-28:
On a macOS 10.15.4 catalina install environment i grabbed /bin/nohup and /usr/bin/open and placed them on my opencore usb boot stick.  

Then i used that opencore usb boot stick to bring up the macOS 10.15.4 recovery environment and try both options for realtek usb wifi adapter StatusBarApp in way that would allow it to stay running / displayed on status bar when i exited terminal and went back to main macOS recovery options display.

Both produced the errors shown below.  So it would seem running app to establish network connection and then having it dropped from status bar when you kill terminal and go back to macOS recovery options display is the experience one has to expect to use RealTek usb wifi adapter vs wired ethernet when doing opencore macOS recovery install process.

``` 
pushd /Volumes/BOOT/usrbin
./nohup /Volumes/BOOT/statusbarapp Folder/StatusBarApp.app/Contents/MacOS/StatusBarApp & 
````
```
### stdout ###
-bash: nohup: command not found
[1]+ Exit 127 nohup /Volumes/BOOT/statusbarapp Folder/StatusBarApp.app/Contents/MacOS/StatusBarApp
```
```
### nohup.out contents ###
nohup: can't detach from console: Undefined error: 0
```
```
pushd /Volumes/BOOT/usrbin
./open /Volumes/BOOT/statusbarapp Folder/StatusBarApp.app
``` 
```
### stdout ###
LSOpenURLsWithRole() failed with error -610 for the file /Volumes/BOOT/statusbarapp Folder/StatusBarApp.app.
```

chromer030, 2020-03-29:
Thank you very much for reporting.
I will try to discover new ways and shall reports here.
Thank you again.

myusrn, 2020-03-29:
Thanks for your efforts.  To clarify it doesn't appear critical that the StatusBarApp stays running since once its been loaded and a network selected and password entered that process can die and the wifi network connection enabled by kexts will keep operating on the selected network.  I'll leave it to you to close this thread as i don't think it relates specifically to the driver you are maintaining with this repo but maybe some aspect of the scenario does.

chromer030, 2020-03-29:
It's hopeful that connection remains fully functional even after termination of StatusBarApp helper application , but it will be great if we found a way to keep it running.
I will report any success efforts here.
Thank you.

Mieze, 2020-07-13:
Off-topic!

closed 2020-07-13 by Mieze