Mieze/RTL8111_driver_for_OS_X #28: not working on MSI Z370-A

waini1110, 2020-05-07:
Thanks for your driver. 
The 2.3.0d10 not working on MSI Z370-A, if possible, could you please check it? 

if you need more error informations, please let me know how to check. 

Thanks again.

khoahoc0508, 2020-05-18:
I have same problem too. Please help. Thanks !
@waini1110 can you send me efi efi folder ? i used to Clover all the time and now want to try OpenCore.

Mieze, 2020-05-18:
According to your problem description I'm almost 100% sure that this is another case of PEBKAC. Please fix your system because the driver is not the problem.

waini1110, 2020-05-19:
@Mieze  Thanks for your reply, When I changed from automatic configuration to manual configuration in network settings and forced it to 100M, it started working.
I'm sorry to bother you, have a nice day. 

I will close this issue. 

Thanks again.

closed 2020-05-19 by waini1110