Mieze/RTL8111_driver_for_OS_X #40: Realtek 8118 Gaming no ethernet on V2.3.0

qyzdo, 2020-09-10:
Hello, I upgraded kext from V2.2.2 to V2.3.0 on my Realtek 8118 Gaming on Gigabyte Z390 UD and ethernet stopped working. It was showing no connection all the time, I spent few hours to figure what is going on, and when I downgraded kext to V2.2.2 everything started working again.

Mieze, 2020-09-11:
In case you need help, you should give me a little bit more information because I'm not a clairvoyant. Have you checked the kernel logs or did you spent these hours smoking dope and drinking wine?

qyzdo, 2020-09-11:
I tought it was opencore fault so I was just switching different files and hoping that it will work, then I switched this kext and it started working without problems. I can replace this kext again with the newest version and check everything for you, but you need to tell me how to check logs and other stuff which will tell you more information :)

Mieze, 2020-09-11:
You'll find detailed instructions in post#1 of the driver's thread on insanelymac.com.

closed 2020-09-20 by Mieze