Mieze/RTL8111_driver_for_OS_X #43: RTL8111H not working on new release 2.3.0

thefat32, 2020-09-14:
Hi! First of all, thank you! I've been using ur kext since my first hackintosh!

I updated my build to use your latest release and encountered my Ethernet was not working after update.

My specs:

- MSI H310M PRO-VDH PLUS with RTL8111H Gigabit LAN controller (info from [specs site](https://latam.msi.com/Motherboard/H310M-PRO-VDH-PLUS/Specification))

Kernel [kernel.log](https://github.com/Mieze/RTL8111_driver_for_OS_X/files/5219694/kernel.log) from `log show --predicate "processID=0" --debug` using DEBUG kext version

Netstat info [netstat.txt](https://github.com/Mieze/RTL8111_driver_for_OS_X/files/5219673/netstat.txt) from `netstat -s`

System Preferences Network:

[image]

Will continue to use 2.2 release, please ask me if you need more info about my system or any other info I could provide.

Mieze, 2020-09-14:
Looks like an issue with auto-negotiation. Try to set the medium manually or, in case you already selected the medium manually, please switch back to auto-negotiation. Let's see if it resolves the issue?

dimitarnestorov, 2020-09-27:
This fixed it for me on a GA-H81M-D3H

closed 2020-09-30 by Mieze

atshoom, 2020-10-02:
Hi, Gigabyte Z490-UD (with its RTL8118) with brand new Clean install of 10.15.7 here.

Just to report I had the same auto negotiation issue to get past the "cable unplugged" error.

I had to go in `Advanced...` , `Hardware` tab, -> manually [OK][Apply]
still nothing
I had to go back in `Advanced...` , `Hardware` tab, -> 100baseTX [OK][Apply]
 then I got connectivity.
which is strange cause I`m pretty sure the computer is connected to a 1Gb port on the other side but if I manually selecting 1000 it says the cable is disconnected.

log for reference : 
```
% log show --predicate "processID=0" --debug | grep '(RealtekRTL8111'
2020-10-02 16:29:51.592652-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: EEE support enabled.
2020-10-02 16:29:51.604077-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: RxPoll support enabled.
2020-10-02 16:29:51.604238-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: TCP/IPv4 segmentation offload enabled.
2020-10-02 16:29:51.615991-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: TCP/IPv6 segmentation offload enabled.
2020-10-02 16:29:51.616153-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: TCP/IPv6 checksum offload enabled.
2020-10-02 16:29:51.616477-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Version 2.3.0 using interrupt mitigate value 0x5f51. Please don't support tonymacx86.com!
2020-10-02 16:29:51.616664-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Disable PCIe ASPM.
2020-10-02 16:29:51.643553-0700 0x116      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: RTL8168H/8111H: (Chipset 29), 18:c0:4d:1c:96:f5
2020-10-02 16:35:39.739719-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link up on en0, 100-Megabit, Full-duplex, No flow-control
2020-10-02 14:36:00.038348-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link down on en0
2020-10-02 14:36:23.160700-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link up on en0, 100-Megabit, Full-duplex, No flow-control
2020-10-02 15:01:09.174923-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link down on en0
2020-10-02 15:01:19.764774-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link up on en0, 10-Megabit, Full-duplex, No flow-control
2020-10-02 15:01:28.152260-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link down on en0
2020-10-02 15:02:10.668529-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link up on en0, 100-Megabit, Full-duplex, No flow-control
2020-10-02 15:02:18.421976-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link down on en0
2020-10-02 15:03:08.359209-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link up on en0, 100-Megabit, Half-duplex, No flow-control
2020-10-02 15:03:24.919062-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link down on en0
2020-10-02 15:03:26.494590-0700 0x14e      Default     0x0                  0      0    kernel: (RealtekRTL8111) [RealtekRTL8111]: Link up on en0, 100-Megabit, Full-duplex, No flow-control
```

hunn1, 2021-08-02:
Gigabyte Z490-UD +1