Mieze/RTL8111_driver_for_OS_X #44: realtek rtl8111 v2.3.0 not working no internet connection

bisaalz, 2020-09-15:
I have latest catalina installed but the new kext is not working but rtl8111 v2.2.2 is working fine. I tried everything i remove network connection from preference, restart the system still no internet connection. Please fix this problem.

Mieze, 2020-09-15:
>  I tried everything

These are the words which usually indicate that it's another case of PEBKAC.

bisaalz, 2020-09-15:
> > I tried everything
> 
> These are the words which usually indicate that it's another case of PEBKAC.

really ok lol then please test in different laptop. I am sure i am not the only one has this problem..

bisaalz, 2020-09-15:
> > > I tried everything
> > 
> > 
> > These are the words which usually indicate that it's another case of PEBKAC.
> 
> really ok lol then please test in different laptop. I am sure i am not the only one has this problem..

I read all your instruction