Mieze/RTL8111_driver_for_OS_X #45: RTL8111H not working #2.3.0

SonerOd, 2020-09-19:
Ethernet is seen on System prefrences  but It says cable is not connected. When I trying on windows, it's working properly.

King-stark, 2020-10-02:
I ran into the same problem when I tried to replace from 2.2.2 to 2.3.0.

King-stark, 2020-10-02:
> I ran into the same problem when I tried to replace from 2.2.2 to 2.3.0.
The 2.3.0 driver has been loaded, but the Ethernet en0 interface cannot detect my network access (my environment is big sur beta 9)

SonerOd, 2020-10-02:
I'm back to 2.2.2 now it works. but not working with 2.3.0. Issue is not solved.

Mieze, 2020-10-02:
As nobody has provided any information about what might be wrong, I assume that these are cases of PEBKAC. I'm not a full service provider!

closed 2020-10-02 by Mieze