Mieze/RTL8111_driver_for_OS_X #48: Adopting MacKernelSDK

vit9696, 2020-09-21:
Hi, we published a standalone kernel SDK — [MacKernelSDK](https://github.com/acidanthera/MacKernelSDK) allowing one:
- to target pretty much any macOS kernel version given an adequate target os
- to avoid various deprecations  or DriverKit mess introduced by newer Xcode releases
- to use standard headers for private stuff like the extra methods in IONetworkingFamily

I adopted it in our variant of IntelMausi with [kernel debugging](https://github.com/acidanthera/IntelMausi) and I wonder if you could find it useful for all your network drivers. Let me know if you need further assistance.

a250995, 2020-10-03:
Latest version did not works,i tried this morning

Mieze, 2020-10-05:
Sounds interesting but at the moment I have no time for hackintoshing at all. Will take a look at it later!

a250995, 2020-10-05:
I got it,try to check what's issues after your work,best wish

aliakbarmostafaei, 2020-10-07:
Medium Auto-negotiation stopped working on 10.15.7 and OC 0.6.2, as indicated by printing _"No medium selected. Falling back to autonegotiation."_ in the logs. @vit9696 would be willing to fork this and adopt MacKernelSDK and other features in your own drivers?

Mieze, 2020-10-07:
@aliakbarmostafaei First of all, sorry for being so harsh, but are you aware of the fact that you are talking nonsense? This log line is normal and doesn't mean that there is something wrong.

Second, due to the number of users with RTL8111H reporting problems, I ordered an RTL8111H from China which arrived this morning. I put it into my machine, installed the driver and it's working on perfectly 10.15.7. There is nothing wrong with the driver, it's working as expected. In case you've got issues, please fix your system or your hardware but it's definitely not the driver which is responsible for your problem.

aliakbarmostafaei, 2020-10-07:
@Mieze My connection drops as soon as this line is printed! And I only updated to OC 0.6.2 with no configuration changes!
Obviously you don't have time or willingness to update this driver as you mentioned in the comments above, so why in the world you take offense if someone took over and fixed it?!
If many people are reporting problems then chances are this driver is broken!! And it only gets fixed if you were more open to suggestions and criticism, one example is the OP's suggestion which you dismissed by having no time, so you are in no position to disrespect people for suggesting fixes on an open source software.

vit9696, 2020-10-07:
@aliakbarmostafaei please do refrain from posting unrelated stuff to the issues in the bugtracker or mentioning me in regards to your personal requests. By arrogantly angering people here you essentially make your problem to never be resolved.

For us this driver works just fine as for Laura. If it fails for you, it could indeed be faulty hardware or a specific issue with your chip revision that is not reproducible on the majority of other chips. In any case, the driver codebase is fully opensource, which means that nothing prevents you from investigating the issue on your own and contributing a patch. This would definitely be more productive than indulging into non-technical polemics.

Lastly, the need to create a standalone codebase for IntelMausi was mainly dictated by our needs in kernel debugging and the support for legacy operating systems. The quality of the changes we made is not really suitable for upstream, so we keep a driver separate at this point.

aliakbarmostafaei, 2020-10-07:
@vit9696 wow! You guys are amazing! Commenting on your own suggestions makes you angry?! This is getting nowhere, we're done here but do the community and yourselves a favor and spend some time on your mental health as you do on your softwares. Not a word in my comments were even remotely offensive!
Peace.

Mieze, 2020-10-07:
@aliakbarmostafaei  I have spent a lot of time writing and testing version 2.3.0 and a lot of competent users tested the development versions I used to publish on insanelymac.com without finding any issue. I've checked the driver's source code several times after the first problem reports and I'm also studying the IONetworking.kext source code whenever Apple releases an update with the result that there is no reason to assume that something has to be done.

I'm not willing to fix the driver because there is nothing to fix! If users are reporting issues with the RTL8111H it's because this is the Realtek NIC which can be found on any mainboard or add-in card using a Realtek chip these days.

If you are not smart enough to to make it work or too lazy to fix your system, don't blame me because I'm not responsible for other people's incompetence or lazyness. The fact that you are posting here shows that you have failed to read the instructions because support requests should be posted in insanelymac.com

Please do me a favor and go to /dev/null or anywhere else but stop wasting my time with your moaning.

aliakbarmostafaei, 2020-10-07:
@Mieze Please stop mentioning me in your comments, as I said we're done here.
FYI I did read and follow the instructions multiple times but to no avail! And then I did post to your insanelymac.com thread along with the kernel logs and config.plist and after 18 hours I have yet to receive a response, now who's being lazy?! 😄 You only commented here because you took offense which by the way I did NOT mean at all. Good luck with your work.

kocoman2, 2022-11-13:
Is there any update? I look at the applertl8139 sources on the receivepacket part but its quite different because it defines some bit stuff .. I have try applertl8169 kdp works but is not usable (disconnects easily).  I look at receivepacket part in aerror2 build but its very different too

thx

closed 2025-02-05 by Mieze