Mieze/RTL8111_driver_for_OS_X #51: Crash caused by sleep during file download

kristinaking, 2020-10-19:
panic(cpu 8 caller 0xffffff801ade9996): Kernel trap at 0xffffff801ebc936a, type 14=page fault, registers:CR0: 0x000000008001003b, CR2: 0xffffffa0aa2354f0, CR3: 0x0000000026db6000, CR4: 0x00000000003406e0RAX: 0x0000000000000000, RBX: 0xffffff937e920000, RCX: 0x00000000000004f0, RDX: 0x0000000000000100RSP: 0xffffffa0afa9bca0, RBP: 0xffffffa0afa9bd30, RSI: 0xffffff937e97e080, RDI: 0x000000000000004fR8:  0x0000000000000000, R9:  0x0000000000000433, R10: 0x197ae7ff73d6b391, R11: 0x0000000000000000R12: 0xffffffa0aa235000, R13: 0xffffff937e920000, R14: 0xffffff937e920000, R15: 0x0000000000000000RFL: 0x0000000000010216, RIP: 0xffffff801ebc936a, CS:  0x0000000000000008, SS:  0x0000000000000000Fault CR2: 0xffffffa0aa2354f0, Error code: 0x0000000000000000, Fault CPU: 0x8, PL: 0, VF: 1Backtrace (CPU 8), Frame : Return Address0xffffffa0afa9b6c0 : 0xffffff801acb666d 0xffffffa0afa9b710 : 0xffffff801adf8f93 0xffffffa0afa9b750 : 0xffffff801ade95ca 0xffffffa0afa9b7a0 : 0xffffff801ac5ba2f 0xffffffa0afa9b7c0 : 0xffffff801acb5f0d 0xffffffa0afa9b8e0 : 0xffffff801acb61f8 0xffffffa0afa9b950 : 0xffffff801b4b792a 0xffffffa0afa9b9c0 : 0xffffff801ade9996 0xffffffa0afa9bb40 : 0xffffff801ade967d 0xffffffa0afa9bb90 : 0xffffff801ac5ba2f 0xffffffa0afa9bbb0 : 0xffffff801ebc936a 0xffffffa0afa9bd30 : 0xffffff801ebc9128 0xffffffa0afa9bd50 : 0xffffff801d371360 0xffffffa0afa9bd80 : 0xffffff801b4137df 0xffffffa0afa9bde0 : 0xffffff801d3713d8 0xffffffa0afa9be40 : 0xffffff801d37577b 0xffffffa0afa9bec0 : 0xffffff801b352d7a 0xffffffa0afa9bfa0 : 0xffffff801ac5b13e       Kernel Extensions in backtrace:         com.apple.iokit.IONetworkingFamily(3.4)[59424AFD-0C6E-33B7-B41C-CDC6D8BE7A1A]@0xffffff801d366000->0xffffff801d37cfff         com.insanelymac.RealtekRTL8111(2.3)[E288C67E-D8FB-3A98-A314-DF21F2B3534B]@0xffffff801eb56000->0xffffff801ebd6fff            dependency: com.apple.iokit.IONetworkingFamily(3.4)[59424AFD-0C6E-33B7-B41C-CDC6D8BE7A1A]@0xffffff801d366000->0xffffff801d37cfff            dependency: com.apple.iokit.IOPCIFamily(2.9)[CD9AB596-4346-3922-B275-92579AB71D07]@0xffffff801d5ef000->0xffffff801d616fffProcess name corresponding to current thread: kernel_taskBoot args: npci=0x2000 alcid=1 Mac OS version:20A5395gKernel version:Darwin Kernel Version 20.1.0: Wed Oct  7 21:17:40 PDT 2020; root:xnu-7195.40.137.201.1~1/RELEASE_X86_64Kernel UUID: D5AE727A-F8F2-31EE-B874-4E907B0C5115KernelCache slide: 0x000000001aa00000KernelCache base:  0xffffff801ac00000Kernel slide:      0x000000001aa10000Kernel text base:  0xffffff801ac10000__HIB  text base: 0xffffff801ab00000System model name: iMacPro1,1 (Mac-7BA5B2D9E42DDD94)System shutdown begun: NOPanic diags file available: YES (0x0)Hibernation exit count: 0System uptime in nanoseconds: 544105591747Last Sleep:           absolute           base_tsc          base_nano  Uptime  : 0x0000007eaf3877a1  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000  Wake    : 0x0000000000000000 0x00000013d71fbd8c 0x0000000000000000
--

Mieze, 2020-10-19:
Thta's a linker error. Clean system caches in order to resolve this as described in the driver's thread on insanelymac.com.

closed 2020-10-19 by Mieze

kristinaking, 2020-10-19:
Clean system cache does not work