Mieze/RTL8111_driver_for_OS_X #53: Ping timeout / packet loss

silvanverschuur, 2020-10-21:
I'm using Catalina (10.15.7) on a MSI Z77MA-G45 motherboard with an integrated Realtek 8111E LAN controller.

I've got an issue regarding ping timeout / packet loss. When I perform a ping to another machine in my network I get about 3% to 4% timeouts / packet loss. In the past I've used version 2.2.1 but recently I've updated the kext to version 2.3.0. The behaviour is the same with both versions. I tried to disable Energy Efficient Ethernet but that didn't help.

I'm quite sure that it's a software issue because Windows 10 (dual boot) doesn't have this issue. 

Do you have any other suggestions or is this a common problem?

Mieze, 2020-10-22:
Does the driver operate properly under heavy load? In case of yes, it's most likely a power management issue and I wouldn't be surprised if it's ASPM related. ASPM is PCIe power management method putting the link between the PCH and the PCIe device into a low power state while in idle mode. Both PCIe ports (the NIC and the PCH root port) always have to use the same configuration or you'll get unpredictable results. Read accesses may return garbage values and write accesses might be discarded resulting in erratic behavior of the NIC.

ASPM doesn't work properly on many hackintoshs because of several reasons:
- ASPM  support is broken on some members of the RTL8111 family.
- The BIOS may enable or enable ASPM. It might leave control of ASPM to the OS or lock the ASPM setting making it impossible for the OS to control ASPM.
- This is the reason why the driver tries to disable ASPM on the PCH's and on the NIC's port but in case the BIOS has enabled it and locked the setting, there will be trouble.

Take a look at your BIOS setting and check if there is anything about ASPM. If possible, leave ASPM control to the OS, otherwise disabled it.

silvanverschuur, 2020-10-22:
The driver works fine under heavy load. I have used iperf to test the performance between my hackintosh and another Linux computer in my network. Most of the time I got speeds > 900 Mbit/sec. I've never had any issues downloading large files or anything like that. 

I think you're right that it has something to do with power management. Ping never fails during performance testing....only when my machine is idle.

I've checked my Bios but I don't have an ASPM setting. Hackintool tells me that ASPM is disabled for all devices. I also tried to disable ASPM in clover (globally and using a device property 'pci-aspm-default') but that also didn't help.

For now I'm out of options so maybe I should leave it like this and ignore the ping timeouts. Thanks for your help anyway!

closed 2020-10-28 by Mieze