Mieze/RTL8111_driver_for_OS_X #55: 2.3.0 Doesn't work on my computer (and many others). Had to revert to 2.2.0

BitGrub, 2020-11-11:
It's a (little?) known issue that this driver has been broken since august. So if anyone else is having issues, try the older version

Mieze, 2020-11-11:
One more case of PEBKAC. Please fix your system instead of talking nonsense!

closed 2020-11-11 by Mieze

BitGrub, 2020-11-11:
System's fine. The public release version of this is known to be broken, sorry didn't mean to offend you

BitGrub, 2020-11-11:
Also hiding this issue is just going to cause more headaches for people troubleshooting 🤦‍♀️

BitGrub, 2020-11-11:
@Mieze

Mieze, 2020-11-11:
I don't care for these people at all. If they aren't smart enough to make it work as most users do, they might want to get a real Mac or go to /dev/null. I'm getting tired of wasting my time with brain dead people.

BitGrub, 2020-11-11:
Alright, no need to be so sour. Thanks for your work, get well soon

BitGrub, 2020-11-11:
@Mieze 
No hard feelings, I completely understand your frustration with dealing with morons. I just thought having this issue listed on the github page might reduce their numbers a bit? Also you can't call me "not smart enough to make it work" when I clearly indicated I am IN THE TITLE

In any case, if you ever want to investigate the issue with your driver, I have a Realtek 8111H Gigabit LAN controller in my MSI b350m mortar motherboard, and would gladly give any more relevant information if you need it. And by not working I mean there's no internet connection, can't even ping anything, but MacOS accurately identifies my ethernet controller.

Have a good one

Mieze, 2020-11-11:
I already spent a lot of time investigating the issue to no avail. I even bought an add-in card with an RTL8111H in order to test it myself and I can confirm that it's working fine. Besides that, if there really was a bug, somebody would have sent me a substantial bug report which would have given me a starting point for further investigations but up to now, I haven't received anything . Instead of that, the only thing I got were repeated laments "it doesn't work". At a certain point you to make a decision: go crazy or stop listening for these cries.