Mieze/RTL8111_driver_for_OS_X #56: v2.3.0 doesn't work sometimes

wujekbogdan, 2020-11-13:
I have an Asus PRIME Z490-P motherboard with Realtek RTL8111H (let's call it **PC A**). I installed the most recent `2.3.0` driver. I see that the kext is loaded, I can see the RTL8111H being recognized by the system (it shows up under System Report and has a mac address assigned), but the ethernet port doesn't light up when I connect the cable.

So the first thought was that something is wrong with the hardware (router, cable or the Realtek itself). 

- I connected a different PC to the same router with the same cable (let's call it **Router A**): it worked fine.
- I booted my **PC A** using Ubuntu live cd: it worked fine on **Router A**

So it looked like there is something wrong with the kext itself.  But just out of curiosity I connected **PC A** to another router - let's call it **Router B**. The ethernet port lighted up properly when I connected the cable. 

It started to look even weirder, so I connected another PC (it's a PC running Linux, let's call it **PC B**) to **Router A** - It also worked fine.

So I did one more test. I downgraded RTL8111H to `2.2.2` and connected it again to **Router A**. Surprisingly it started to work fine on **Router A**!

So wrapping things up:
- **PC A** with v2.3.0 works on **Router B*, but does not work on **Router A**
 - **PC A** with v2.2.2 works on  both routers
 - **PC B** works on both routers

So it looks like there's some regression in `v2.3.0` that causes the issue.

---
I know that this post is a bit long and does not provide any information that would help you with debugging. Is there anything I can provide you with? Can I grab some logs from v2.3.0?

bisaalz, 2020-11-14:
> I have an Asus PRIME Z490-P motherboard with Realtek RTL8111H (let's call it **PC A**). I installed the most recent `2.3.0` driver. I see that the kext is loaded, I can see the RTL8111H being [...]

Hey can u do this go to setting->network click your  ethernet on your left side->chose advanced->go to hardware tab->select configure manually->speed 100basetx,duplex->full duplex, mtu->standard(1500) click apply and then go out and check you should get ethernet i solved by this method in v2.3.0.

wujekbogdan, 2020-11-14:
Thanks, this solves the problem. Still, this seems to be a bug because this should work out of the box with auto settings.

Mieze, 2020-11-14:
The driver is based on Realtek's Linux driver and it applies patches to the NIC's firmware which are included in the source code provided by Realtek. Version 2.3.0 is based on an newer version of the Linux driver than 2.2.2 because it adds support for new members of the RTL8111 family and it might also come with modified firmware for older chips. This could also be the reason why some users are experiencing problems with 2.3.0.

Unfortunately these firmware patches are black box code and there is nothing I can do in order to resolve this. Only Realtek knows what they are doing and the only thing I can do is to hope that they may take care of the issue.

Apart from this I don't see it as a problem if you have to select the medium manually. There are ethernet devices from several manufacturers and each of them has certain peculiarities caused by different implementations of the standard. Auto-negotiation should result in a stable connection but it's not guaranteed that it will negotiate a connection with optimal parameters so that it may be the better solution to select the medium manually in certain cases. I've also seen such behavior with NICs from other vendors so that I don't thing that it's something I should worry about.

closed 2020-11-14 by Mieze

wujekbogdan, 2020-11-15:
I had to revert to `v2.2.2`. Manual settings stop working after the system woke up from sleep (it doesn't happen always, so far it happened just once). Changing the settings to auto and then setting them to manual didn't help. 

I'm not sure is it an issue that's specific to my system or is it the driver that causes the problem, but reverting the kext to old version solves the problem.