Mieze/RTL8111_driver_for_OS_X #58: Very low download Speed

anubhaavspark, 2020-12-10:
I have latest version of Kext installed. But download speed is too low, on the other hand upload speed is good and as accepted. My plan is 300 MBPS. I have attached the screenshot.

[image]

nikhil19rathore99, 2020-12-13:
in my laptop with BigSurr it doesn't works.
help!!

mehmetyukcell, 2020-12-23:
> in my laptop with BigSurr it doesn't works.
> help!!

Please, system preferences, networks, ethernet, advanced level, 
[image]

Mieze, 2021-01-15:
Please read the instructions because you haven't. Otherwise you wouldn't post support requests here on GitHub.

closed 2021-01-15 by Mieze

anubhaavspark, 2021-01-16:
My original request was for "Slow Download Speed" in BigSur. It worked perfectly in Catalina. So, yes I have read instructions.