Mieze/RTL8111_driver_for_OS_X #59: Not working in BigSurr 11.0.1

nikhil19rathore99, 2020-12-13:
It would not working in my hack book works well in Catalina 
Please help!!
Thanks for your valuable time.

norbertschuler, 2020-12-13:
On my MacBook Pro with macOS Big Sur it works. But I had to partially disable SIP in [Recovery mode](https://osxdaily.com/2017/05/18/access-terminal-recovery-mode-mac/) with `csrutil disable` and `csrutil enable --without kext` (https://developer.apple.com/forums/thread/17452) so in the end it looks like this
```
$ csrutil status
System Integrity Protection status: unknown (Custom Configuration).

Configuration:
	Apple Internal: disabled
	Kext Signing: disabled
	Filesystem Protections: enabled
	Debugging Restrictions: enabled
	DTrace Restrictions: enabled
	NVRAM Protections: enabled
	BaseSystem Verification: enabled

```

weslleystos, 2020-12-19:
Hi, for me doesn't work after version 2.2.

Aorus b450 Elite, 
Ryzentosh BigSur.

chromer030, 2021-01-14:
For me also doesn't work after version 2.2 , tried every guide , fresh installation but no success.
Only version 2.2 working properly.

Intel Skylake.
macOS BigSur

gigadata, 2021-01-14:
Not Connection Ethernet, its problem etherner Disabled and Enabled automatic,
my hardware
Ryzen 5 3600 
msi b450m pro-vdh plus
Realtek® 8111H Gigabit 

Can Help me?

Mieze, 2021-01-15:
I can confirm that the driver is working properly with Big Sur after running a lot of tests. In case you've got problems, please read the instructions and fix your system. Posting support requests here on GitHub clearly shows that you haven't.

closed 2021-01-15 by Mieze