Mieze/RTL8111_driver_for_OS_X #60: Minimum macOS/kernel requirements

1alessandro1, 2021-01-27:
I'm planning to test Snow Leopard (10.6.7 10J4139) on an old machine (i3 540 + GT 320 OEM) which already has Catalina (with a GT710) with 10.15.7 19H115 fully functional with this excellent kext, (Darwin 19.6.0)

What are the official minimum kernel/macOS requirements with this kext? Is Snow Leopard 10.6.7 comatible or should I check the older releases on InsanelyMac for this kext? 

Thanks

1alessandro1, 2021-01-27:
Update:

I've read the`ReadMe.md` which says *Snow Leopard requies some changes.* 

Could you provide some additional info or a prebuilt if it's not too time consuming? 
Thanks again.

Mieze, 2021-01-28:
AFAIK RehabMan made a version of the driver for Snow Leopard years ago. It's was derived from version 1.x.x because starting with version 2.0.0 the driver uses a private driver interface which was introduced with 10.8.5.

As I removed the 1.x.x build target together with the corresponding code, you'll have a hard time adding it again to newer versions of the driver but provided you only need support for the RTL8111F and below, RehabMan's driver might be the way to go.

1alessandro1, 2021-01-29:
> RehabMan's driver might be the way to go.
Does [this](https://bitbucket.org/RehabMan/os-x-realtek-network/downloads/RehabMan-Realtek-Network-2014-0523.zip) build support this network card?

[image]

forget about the GT710, I'm going to use a GT320 for SnowLeopard

Mieze, 2021-01-31:
Although it's impossible to identify different versions of the RTL8111 by it's PCI dev ID because all family members share the same one, the fact that it's mounted on a 5 series board makes me assume that it's an RTL8111F or below so that it should be supported by any version of the driver.

1alessandro1, 2021-01-31:
Thanks for your help, I'm going to test the latest 1.x.x as soon as possible, closing

closed 2021-01-31 by 1alessandro1