Mieze/RTL8111_driver_for_OS_X #61: Request for RTL1836 driver

MiddlemanTM, 2021-01-27:
Hi Mieze,

Have been a fan of your work for years! Wanted to ask you if it is possible to create a new driver as I'm trying to run High Sierra 10.13.6 on my system. I have a HP Pavilion 17 laptop with a i7-7500U and it has a Realtek RTL1836 card fitted internally. The closest kext I've found is a PCGenRTL1839Ethernet.kext but it doesn't work, nor does the current RTL8111 drivers afaik. Much appreciated if you could help with this.

Mieze, 2021-01-27:
You are in the wrong project! 8136 is the PCI dev ID of the RTL8100 family members. Try this driver!

closed 2021-01-27 by Mieze

MiddlemanTM, 2021-01-28:
Hi Mieze,

I must apologize - I mistyped the model number in the header - it was RTL8136 that I was meant to refer to. I have since tried your 8111 driver as mentioned and unfortunately it did not respond nor show up in the System Profiler or under System Preference > Network drivers. That was why I posted the request.

chromer030, 2021-01-28:
@MiddlemanTM 
Did you tried V2.2.2 ?
Try it , it works !