Mieze/RTL8111_driver_for_OS_X #63: Jumbo MTU problem

coldcue, 2021-02-16:
I have a Jumbo frame compatible network chip on my GA Z390 UD motherboard and the driver does not recognize the jumbo frame option.

Hackintool output:
`06:00.0 10EC 8168 1458 E000 Disabl Realtek Semiconductor Co., Ltd RTL8111/8168/8411 PCI Express Gigabit Ethernet Con Network controller   Ethernet controller  P03@1C,2/PXSX@0 ethernet        PciRoot(0x0)/Pci(0x1C,0x2)/Pci(0x0,0x0) `

Mieze, 2021-02-16:
User error -> read instructions!

coldcue, 2021-02-16:
> User error -> read instructions!

Oops, yep I'll then post on insanelymac.

Although with all respect to you by creating this and making it available for us, but I'm also a dev and I think Github as a platform for open source stuff with their new issues feature is very neat. If I can have an advice to you that it's worth a try.

closed 2021-04-29 by Mieze