Mieze/RTL8111_driver_for_OS_X #71: The driver just dies

TheMorc, 2021-09-01:
Hi.

GA-Z97-HD3, i5-4460, GTX 650, BIg Sur 11.1

I'm using this driver for quite some time already, it's just fine except for that it dies when you load it. You then need to disconnect and connect the RJ-45 back into the motherboard.
Everything was fine on Yosemite, El Capitan, High Sierra, Mojave, Catalina, and Big Sur betas until i installed Big Sur in December 2020. It just dies there when i start transferring files or when i use NDI between machines. 

[rtl8111.log.zip](https://github.com/Mieze/RTL8111_driver_for_OS_X/files/7092660/rtl8111.log.zip)

diab303, 2021-11-06:
I have a similar problem with a H97 board under Mojave (driver sometimes dies when there is heavy network load then need to re-plug the cable to fix it) - changing the Duplex setting under Network/Hardware seemed to improve the situation, maybe it will help you, too. It seems to also depend on what sort of switch it's plugged into (maybe because of auto-negotiation or extra ethernet features like flow control or energy efficiency), I have a tp-link switch that never fails but directly plugging the PC into the cable modem triggers this issue once or twice a day.

TheMorc, 2021-11-06:
I bought a USB ethernet adapter since then + i'm planning to buy one of those 16" MacBooks since having a desktop is not that great for me anymore after all of those years.

Mieze, 2021-11-06:
Please read instructions!

closed 2021-11-06 by Mieze