Mieze/RTL8111_driver_for_OS_X #75: Not working on Ventura beta

rul3v01d3r, 2022-06-18:
[image]
Not working on Ventura beta, but kext is loaded and pic device is shown  in Hackintool. 
[image]
[image]
[image]

Mieze, 2022-06-20:
First, this is the wrong place to post. Second, what does an USB device have to do with the driver? Check your system and/or post in the right place.

closed 2022-06-20 by Mieze

rul3v01d3r, 2022-06-23:
> First, this is the wrong place to post. Second, what does an USB device have to do with the driver? Check your system and/or post in the right place.

i m showing the ethernet devices not the usb device and i thought this is the place to open the issue if not i m sorry for that, but i want to show u that ur kext is loaded but not working on ventura beta
my specs are:
cpu 5800x
mobo msi x570 a pro
ethernet realtek8111/8168
os ventura beta
tried ur latest version i.e. 2.4.2

wolter8, 2022-11-16:
> i m showing the ethernet devices not the usb device and i thought this is the place to open the issue if not i m sorry for that, but i want to show u that ur kext is loaded but not working on [...]

Did you solve this? I have the same mono and the same problem, I've tried all versions and nothing working.

rul3v01d3r, 2022-11-16:
for msi u need pci=0x80000000 bootargs and latest patch