Mieze/RTL8111_driver_for_OS_X #81: System reboot after this drive initializes - macOS Sonoma

techflashYT, 2023-07-02:
After this driver initializes, the system immediately reboots, no panic, nothing.
I am using macOS Sonoman, and Linux says that my Ethernet controller is `Realtek Semiconductor Co., Ltd. RTL8111/8168/8411`.

Mieze, 2023-07-02:
Well, this could be anything, staring with an ACPI problem up to a subsystem which depends on the network. Unless you give can give me a KP, there is nothing I can do for you and I doubt that this is a driver issue.

blablagaming, 2023-10-16:
im also having this issue on macOS Monterey.

zearp, 2024-04-22:
Having similar "issues" and can reproduce them. I think this is related to a firmware being present from another OS if that makes any sense. I'm not technical like that but when booting into another OS and then booting back into macOS there's pretty much always an instant reboot as soon this kext is loaded. This also happens with other kexts like the Intel bluetooth one.

I think this is because there's a firmware loaded by another OS and isn't removed/unloaded (if possible) so macOS freaks out and just reboots. Sometimes it keeps bootlooping and sometimes it boots fine on a 2nd or 3rd go. Turning the computer off, removing the cable and holding down the power button always fixes it. I can reproduce it but there's really nothing on screen that would help. I recorded the screen and there's no error or anything, it just reboots. 

In my limited view a fix could be something in OpenCore or a different kext that's loaded very early on that flushes any present firmwares in devices before it continues. I can try the debug version of this kext if the maintainer wants to look into this but personally this doesn't really affect me and the work around is easy. Either don't boot into another operating systems or do a full power down before booting back into macOS.

closed 2025-01-22 by Mieze